raspberrypi/Raspberry-Pi-OS-64bit #232: Latest VLC Upgrade Fail

JBHoren, 2022-07-01:
[horen@pie tmp]$ apt-get dist-upgrade                                                                                                                                               
Reading package lists... Done                                                                                                                                                                
Building dependency tree... Done                                                                                                                                                             
Reading state information... Done                                                                                                                                                            
Calculating upgrade... Done                                                                                                                                                                 
The following packages will be upgraded:                                                                                                                                                     
  libvlc-bin libvlc5 libvlccore9 python3-sense-hat vlc vlc-bin vlc-data                                                                                                                      
  vlc-l10n vlc-plugin-access-extra vlc-plugin-base vlc-plugin-notify                                                                                                                         
  vlc-plugin-qt vlc-plugin-samba vlc-plugin-skins2 vlc-plugin-video-output                                                                                                                   
  vlc-plugin-video-splitter vlc-plugin-visualization                                                                                                                                         
17 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.                                                                                                                              
Need to get 13.0 MB of archives.                                                                                                                                                             
After this operation, 345 kB of additional disk space will be used.                                                                                                                          
Do you want to continue? [Y/n]                                                                                                                                                               
Get:1 http://archive.raspberrypi.org/debian bullseye/main arm64 vlc arm64 3.0.17                                                                                                             .4-0+deb11u1+rpi1+rpt1 [146 kB]                                                                                                                                                              
Get:2 http://archive.raspberrypi.org/debian bullseye/main arm64 vlc-plugin-base                                                                                                              arm64 3.0.17.4-0+deb11u1+rpi1+rpt1 [2,878 kB]                                                                                                                                                
Get:3 http://archive.raspberrypi.org/debian bullseye/main arm64 vlc-data all 3.0                                                                                                             .17.4-0+deb11u1+rpi1+rpt1 [457 kB]                                                                                                                                                           
Get:4 http://archive.raspberrypi.org/debian bullseye/main arm64 libvlccore9 arm6                                                                                                             4 3.0.17.4-0+deb11u1+rpi1+rpt1 [512 kB]                                                                                                                                                      
Get:5 http://archive.raspberrypi.org/debian bullseye/main arm64 vlc-plugin-skins                                                                                                             2 arm64 3.0.17.4-0+deb11u1+rpi1+rpt1 [571 kB]                                                                                                                                                
Get:6 http://archive.raspberrypi.org/debian bullseye/main arm64 vlc-plugin-qt ar                                                                                                             m64 3.0.17.4-0+deb11u1+rpi1+rpt1 [1,102 kB]                                                                                                                                                  
Get:7 http://archive.raspberrypi.org/debian bullseye/main arm64 vlc-plugin-video                                                                                                             -output arm64 3.0.17.4-0+deb11u1+rpi1+rpt1 [204 kB]                                                                                                                                          
Get:8 http://archive.raspberrypi.org/debian bullseye/main arm64 libvlc5 arm64 3.                                                                                                             0.17.4-0+deb11u1+rpi1+rpt1 [175 kB]                                                                                                                                                          
Get:9 http://archive.raspberrypi.org/debian bullseye/main arm64 vlc-bin arm64 3.                                                                                                             0.17.4-0+deb11u1+rpi1+rpt1 [165 kB]                                                                                                                                                          
Get:10 http://archive.raspberrypi.org/debian bullseye/main arm64 libvlc-bin arm6                                                                                                             4 3.0.17.4-0+deb11u1+rpi1+rpt1 [128 kB]                                                                                                                                                      
Get:11 http://archive.raspberrypi.org/debian bullseye/main arm64 python3-sense-h                                                                                                             at all 2.3.1-1 [16.8 kB]                                                                                                                                                        
Get:12 http://archive.raspberrypi.org/debian bullseye/main arm64 vlc-l10n all 3.                                                                                                             0.17.4-0+deb11u1+rpi1+rpt1 [5,918 kB]                                                                                                                                                        
Get:13 http://archive.raspberrypi.org/debian bullseye/main arm64 vlc-plugin-acce                                                                                                             ss-extra arm64 3.0.17.4-0+deb11u1+rpi1+rpt1 [139 kB]                                                                                                                                         
Get:14 http://archive.raspberrypi.org/debian bullseye/main arm64 vlc-plugin-noti                                                                                                             fy arm64 3.0.17.4-0+deb11u1+rpi1+rpt1 [129 kB]                                                                                                                                               
Get:15 http://archive.raspberrypi.org/debian bullseye/main arm64 vlc-plugin-samb                                                                                                             a arm64 3.0.17.4-0+deb11u1+rpi1+rpt1 [130 kB]                                                                                                                                                
[... cut ...]
libEGL warning: DRI2: failed to authenticate
[0000007f080013e0] mmal_xsplitter vout display error: Failed to open Xsplitter:opengles2 module
[0000007f080013e0] mmal_xsplitter vout display error: Failed to open Xsplitter:mmal_vout module
[0000007f2cf53d10] main decoder error: buffer deadlock prevented
[0000007f2cf53d10] main decoder error: buffer deadlock prevented
[00000055bb75f4f0] alsa audio output error: cannot estimate delay: Broken pipe
QObject::~QObject: Timers cannot be stopped from another thread

XECDesign, 2022-07-01:
@popcornmix @jc-kynesim

jc-kynesim, 2022-07-03:
What are you trying to play? Is this in X?

jc-kynesim, 2022-07-03:
Given you are launching from a command line can you do "vlc -vv |& tee t.log" and zip up the log for me?

JBHoren, 2022-07-03:
> What are you trying to play? **Is this in X?**

Yes.

> Given you are launching from a command line can you do 
> "vlc -vv |& tee t.log" and zip up the log for me?

No. In order to regain use of vlc, I restored all of /usr from my 6AM backup (I use rpi-clone) from the SD card (I boot from usb drive). Why? Because I first ran "sudo apt-get remove vlc" to remove vlc-3.0.17; doing so nuked _many_ other files/apps dependent on vlc (digikam, for one... which I use daily). So, I'm now back with a working system (vlc-3.0.16, digikam, etc.), and I'm unwilling to go through all this a second time. Sorry. I ran "sudo apt-mark hold vlc*" to avoid future problems.

JBHoren, 2022-07-04:
OK, forget my previous comment. I changed boot order to SD card, rebooted, then ran "sudo apt-get update" and "sudo apt-get dist-update". vlc is now at 3.0.17.4. As requested, I ran "vlc -vv |& tee t.log" and have attached the results.

[t.log](https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/files/9042269/t.log)

jc-kynesim, 2022-07-05:
I'm sorry, I think I failed to be specific - I wanted a logfile that covered you trying to play something and that log looks like just startup.  If you tried to play something there then I'm really confused.

JBHoren, 2022-07-05:
Yes, but it's OK. In any case, I rebooted from the SD card, again updated to the latest version of vlc, and... audio **and** video. Rebooted from the USB drive, updated to the latest version of vlc, and... audio **and** video.

What's changed?

At my end, nothing. When I first upgraded to vlc-3.0.17.4 and found that the video didn't work, I (stupidly) uninstalled it with "sudo apt-get remove vlc". After realizing that I'd lost digikam and all the rest, I rebooted from SD, mounted USB:/root, and copied SD:/usr onto/over it. Reboot from USB returned my system to status quo ante. 

Obviously, something _has_ changed. I'll take the win.

lurch, 2022-07-06:
> I rebooted from SD, mounted USB:/root, and copied SD:/usr onto/over it.

In general, you shouldn't copy system folders around like that, because it'll cause `apt-get` to get confused about what is or isn't installed on your system (and _may_ possibly lead to further things being broken?).