raspberrypi/Raspberry-Pi-OS-64bit #234: libcamera broken - hangs after "INFO RPI raspberrypi.cpp:1368 Registered camera"

JooJooBee666, 2022-07-07:
Just today, I just setup a test Pi 4B2 I have here with the new Raspberry OS 64 bit "Bullseye".  Before that, I was using picamera to access camera functions and everything was fine.  I setup aq new SD card with Raspberry Pi OS 64 (current) and worked on installing all the libcamera modules for the OS and python.  Initially, I was able to get some results but after installing some additional updates, its now just broken.  Running any of the libcamera-xxxx or trying some simple Python scripts using picamera2 results in the camera hanging after the following:
```
$ libcamera-hello
Made X/EGL preview window
[0:01:34.233116908] [11277]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3700-f30ad033
[0:01:34.281234626] [11421]  INFO RPI raspberrypi.cpp:1368 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media3 and ISP device /dev/media0
```
I've shut it down, updated firmware, checked for more updates (none found) and rebooted countless times, but I'm still stuck.  Any help appreciated.

Further information:
```
$ uname -a
Linux pi5 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux
```
```
$ sudo rpi-eeprom-update
BOOTLOADER: up to date
   CURRENT: Tue 26 Apr 2022 10:24:28 AM UTC (1650968668)
    LATEST: Tue 26 Apr 2022 10:24:28 AM UTC (1650968668)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader/default)
            Use raspi-config to change the release.

  VL805_FW: Dedicated VL805 EEPROM
     VL805: up to date
   CURRENT: 000138a1
    LATEST: 000138a1
```

lurch, 2022-07-07:
ping @davidplowman and @naushir

naushir, 2022-07-08:
I've just tried a fresh build of Raspberry Pi OS 64-bit and gone through an update cycle with sudo apt update.  Everything seems to be working normally for me.

Could you try with a fresh image again please?  It might be that some of your additional updates or changes could have broken things.

JooJooBee666, 2022-07-08:
Everything I installed since update I need and I'll just have to reinstall again.  If that is what caused it, I will end up here again.  Is there a log or some debugging that can be done to see what's causing the hang instead?

naushir, 2022-07-08:
Could you list all the packages that you have installed to help us narrow it down?  It might be worth starting from a clean slate and then installing packages one by one to see which one might break it.

You can also try and add more verbose logging by calling the application with ``LIBCAMERA_LOG_LEVEL=RPI:0`` environment variable set.

JamesH65, 2022-07-08:
Get a new SD card, put latest image on, test. If that doesn't work then we know there is something wrong with the HW. If it works, add back the stuff you need until it stops working, then last thing installed is causing the problem.

JooJooBee666, 2022-07-08:
Ok, thanks guys.  I'll try checking logs first and another SD card second; I have a few spares.  I'll report back

JooJooBee666, 2022-07-11:
I wasn't able to get anything out of the logs.  So I started a new install on a spare SD Card.  I was unable to reproduce this issue after installing mostly the same packages I had before.  So I'm not sure what caused this, unfortunately.  For now, I'm going to close this since I was unable to reproduce.

closed 2022-07-11 by JooJooBee666

JooJooBee666, 2022-08-04:
Just an FYI, I ran in to this again...but jsut as last time, I don't know what package caused it.  I uninstalled all the stuff I thought I recently installed but could not get it functioning again and had to wipe once more.  Rather aggravating...