raspberrypi/Raspberry-Pi-OS-64bit #236: Feature request: Add option for hidden wifi network in setup wizard

adevoss, 2022-08-04:
Booting a pios image for the first time the wizard starts to configure several things.
When configuring the wifi (to connect to an AP) there is no option to connect to a hidden SSID.

Of course this can be done manually after finishing the wizard but you have to know syntax by heart or get them from another setup somehow.

This being part of the wizard would be convinient.

lurch, 2022-08-04:
I suspect you might be looking for [this](https://github.com/raspberrypi-ui/piwiz) repo?

adevoss, 2022-08-05:
Yes, you are right.
This one was the best I could find between all the repos.
It was a bit of a gamble because this issue is not only for the 64bit version of PiOS.

Can you move it please or should I create a new issue in that repo?

lurch, 2022-08-05:
ping @spl237

spl237, 2022-08-05:
I'll put it on the to-do list, but it'll be a low priority - hiding SSIDs is basically futile as security measures go, as anyone with malicious intent can trivially bypass it, so no-one really bothers any more!

We are planning to move the network configuration to use Network Manager, which will make connecting to a hidden network much easier, but that will be once the desktop is running, not in the startup wizard.

adevoss, 2022-08-05:
> hiding SSIDs is basically futile as security measures go

Very true but why not hide it is my point of view.

The reason for me to ask for this feature request is: the (only?) way to connect to a hidden SSID is to edit wpa_supplicant.conf.
Convinient would be a GUI to do this.
The wizard would be the most logical place if you ask me, but another GUI is fine with me.

Now even if you boot PiOS for the first time, finish the wizard and the desktop is running, there is also no GUI option to join a hidden wifi network.

spl237, 2022-08-05:
As above - this will be available from the main desktop GUI when the change to network manager is finished. There will then be no need to edit config files by hand.