raspberrypi/Raspberry-Pi-OS-64bit #238: Raspberry SPI works in 32 bits but on 64 bits is not working

flopezmontero, 2022-09-02:
Hi, newbie here.

For my equipment, I had Debian 9 32 bits installed on my Raspberry Pi which works as expected my SPI port. But after I switched to Debian 11 64 bits, I enabled my SPI port and configured as much as I could my SPI to perform as expected, but it keeps without communication.

Any help around here?

I add the outputs to try to debug this failure. Both cases were performed on same device:

> 32 bits SPI
pi@DRVTST102:~ $ hostnamectl
   Static hostname: DRVTST102
         Icon name: computer
        Machine ID: 7c1be88c4f344d2db4c258de526b9897
           Boot ID: b72f5e715f91440d967ac18160a91eb2
  Operating System: Raspbian GNU/Linux 9 (stretch)
            Kernel: Linux 4.14.79-v7+
      Architecture: arm
pi@DRVTST102:~ $ uname -a
Linux DRVTST102 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux
pi@DRVTST102:~ $ uname -r
4.14.79-v7+
pi@DRVTST102:~ $ ls /dev/spi*
/dev/spidev0.0  /dev/spidev0.1
pi@DRVTST102:~ $ lsmod | grep spi
spidev                 16384  0
spi_bcm2835            16384  0
pi@DRVTST102:~ $ sudo modinfo spi_bcm2835
filename:       /lib/modules/4.14.79-v7+/kernel/drivers/spi/spi-bcm2835.ko
license:        GPL v2
author:         Chris Boot <bootc@bootc.net>
description:    SPI controller driver for Broadcom BCM2835
srcversion:     64E8A93AAE8A9E8C014112F
alias:          of:N*T*Cbrcm,bcm2835-spiC*
alias:          of:N*T*Cbrcm,bcm2835-spi
depends:        
intree:         Y
name:           spi_bcm2835
vermagic:       4.14.79-v7+ SMP mod_unload modversions ARMv7 p2v8 
pi@DRVTST102:~ $ ./spidev_test -D /dev/spidev0.0
spi mode: 0x0
bits per word: 8
max speed: 500000 Hz (500 KHz)
RX | FF FF FF FF FF FF 40 00 00 00 00 95 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF F0 0D  | ......@....�..................�.
pi@DRVTST102:~ $ ./spidev_test -D /dev/spidev0.1
spi mode: 0x0
bits per word: 8
max speed: 500000 Hz (500 KHz)
RX | FF FF FF FF FF FF 40 00 00 00 00 95 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF F0 0D  | ......@....�..................�.

> 64 bits SPI
amedrvdual@raspberrypi:~ $ hostnamectl
   Static hostname: raspberrypi
         Icon name: computer
        Machine ID: c265ca4399944d068b9853d3c57c73d3
           Boot ID: 10f343592ddb4a9aab0c37f1f0a23b47
  Operating System: Debian GNU/Linux 11 (bullseye)
            Kernel: Linux 5.15.61-v8+
      Architecture: arm64
amedrvdual@raspberrypi:~ $ uname -a
Linux raspberrypi 5.15.61-v8+ #1579 SMP PREEMPT Fri Aug 26 11:16:44 BST 2022 aarch64 GNU/Linux
amedrvdual@raspberrypi:~ $ uname -r
5.15.61-v8+
amedrvdual@raspberrypi:~ $ ls /dev/*spi*
/dev/spidev0.0  /dev/spidev0.1
amedrvdual@raspberrypi:~ $ lsmod |grep spi
spidev                 24576  0
spi_bcm2835            24576  0
amedrvdual@raspberrypi:~ $ sudo modinfo spi_bcm2835
filename:       /lib/modules/5.15.61-v8+/kernel/drivers/spi/spi-bcm2835.ko.xz
license:        GPL
author:         Chris Boot <bootc@bootc.net>
description:    SPI controller driver for Broadcom BCM2835
srcversion:     76B9E825743BD9406356BAA
alias:          of:N*T*Cbrcm,bcm2835-spiC*
alias:          of:N*T*Cbrcm,bcm2835-spi
depends:        
intree:         Y
name:           spi_bcm2835
vermagic:       5.15.61-v8+ SMP preempt mod_unload modversions aarch64
parm:           polling_limit_us:time in us to run a transfer in polling mode
 (uint)
amedrvdual@raspberrypi:~ $ ./spidev_test -D /dev/spidev0.0 
spi mode: 0x4
bits per word: 8
max speed: 500000 Hz (500 KHz)
RX | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................................
amedrvdual@raspberrypi:~ $ ./spidev_test -D /dev/spidev0.1
spi mode: 0x4
bits per word: 8
max speed: 500000 Hz (500 KHz)
RX | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................................

Thank you.

pelwell, 2022-09-05:
You are comparing a 32-bit 4.14 kernel with a 64-bot 5.15 kernel. It is much more likely to be the kernel version change than the switch to 64-bit that introduced the problem.

Install a current 32-bit RPiOS image on a spare SD card and test SPI there. If you find that the 32-bit 5.15 kernel behaves the same as the 64-bit version (i.e. it doesn't work in your configuration), open a new issue at https://github.com/raspberrypi/linux/issues. Please include details of the attached SPI hardware.

flopezmontero, 2022-09-05:
Hi Pelwell,

I've tested with kernel 5.15 and have the same issue. But I want to verify 5.15 also with the OS company supports:

[http://comfilewiki.co.kr/en/doku.php?id=comfilepi:raspbian:index]

Meanwhile, how can I obtain the detail of SPI hardware?

lurch, 2022-09-05:
I've no idea what you mean by "with the OS company supports", but Raspberry Pi engineers (and [documentation](https://www.raspberrypi.com/documentation/)) officially only support the latest versions of Raspberry Pi OS downloadable from https://www.raspberrypi.com/software/operating-systems/ .

For support with "ComfilePi" you'll have to ask the producers of that system.

pelwell, 2022-09-05:
> I've tested with kernel 5.15 and have the same issue.

I asked you try _32-bit_ 5.15. Did you? Try to give more complete answers.