raspberrypi/Raspberry-Pi-OS-64bit #239: avahi-daemon not restarting on restart of computer

drmikecooke, 2022-09-04:
This may be due to recent apt upgrade but I've noticed I've not been able to contact other headless rpis through their hostname without restarting avahi-daemon.

I've traced the problem to the system thinking that avahi-daemon has been continuously running when the system (rpi4) has been powered off. I guess during powerdown it doesn't make a suitable note who knows where and thinks it doesn't need to restart the process on powerup.

XECDesign, 2022-11-10:
Could we get a bit more information please?

> This may be due to recent apt upgrade but I've noticed I've not been able to contact other headless rpis through their hostname without restarting avahi-daemon.

Which packages were updated between the time it was working as you expect and when it stopped? (/var/log/apt/history* might be useful in figuring this out)

> thinks it doesn't need to restart the process on powerup.

Are you actually seeing that the process isn't running? If so, what's the output of `sudo journalctl -b0 -u avahi-daemon` ?

What's actually happening? Is the client not trying to resolve the name? Is the daemon not responding? Are there working and non working traces to look at from both client and server sides?

drmikecooke, 2023-01-23:
Sorry, the issue seems to have sorted itself out, presumably through update

closed 2023-01-23 by drmikecooke