raspberrypi/Raspberry-Pi-OS-64bit #240: SSL error for Perl/Mojolicious Apps

bislink, 2022-10-20:
`wget https://spanel.gnani.xyz:44042` gives the following error. 

GnuTLS: Error in the pull function.                                                                                                                         Unable to establish SSL connection.

Apart from this, I cannot access the/any Mojolicious app that is running on a custom port with SSL whereas `https://spanel.gnani.xyz` (apache2) shows no error.

Perl Version 5.32.1 on Raspberry Pi 4 with 8 GB RAM. 

On Brave and Edge, I get error : ERR_CONNECTION_RESET

Reinstalling openssl did nothing.

appreciate help fixing this issue.

pelwell, 2022-10-20:
I get the same result as @XECDesign - it's not a Raspberry Pi-specific issue, and since the browsers also fail it's looking a lot like a server problem.

bislink, 2022-10-20:
`wget https://spanel.gnani.xyz:44042` is not served by apache2 but Perl/Mojolicious

bislink, 2022-10-20:
Error after reinstalling `libmojoicious-perl` for `wget --spider https://spanel.gnani.xyz:44042` : 

```
GnuTLS: The TLS connection was non-properly terminated.                                                                                                     Unable to establish SSL connection.
```

pelwell, 2022-10-20:
All clients seem to agree that https://spanel.gnani.xyz:44042 is not sending any data - what do you expect us to do?

bislink, 2022-10-20:
Same app without SSL seems to work: http://spanel.gnani.xyz:44040

pelwell, 2022-10-20:
Are you actually telling us that spanel.gnani.xyz is a site hosted on a Pi 4?

bislink, 2022-10-20:
Yes. I am running multiple sites on a Pi 4 8 GB RAM with apache2 and some Perl/Mojo apps with custom ports!

pelwell, 2022-10-20:
Read the server log, check your certificate, etc.

bislink, 2022-10-20:
The same certificate works ok for apache2. `https://spanel.gnani.xyz`

pelwell, 2022-10-20:
It doesn't look as though you are alone - I found this similar error (and the Mojolicious repo): https://github.com/mojolicious/mojo/issues/1876

closed 2022-10-20 by pelwell