raspberrypi/Raspberry-Pi-OS-64bit #243: Problem with top menus for Qt6 apps when connected through VNC but not when through xrdp using X11

hamlatzis, 2022-12-31:
I'm connecting to my new pi4 8GB from Windows using VNC (no monitor attached) and in all my Qt6 applications their top menus don't work, but if I built with Qt5 they're fine.

At the beginning I thought it was a Qt problem, but if I connect using Remote Desktop (using xrdp protocol), then both Qt5 & Qt6 apps work fine.

The problem with xrdp is I need to create a second user and cannot use my primary. 

Is there a solution for VNC? If not is there a solution to use the primary user under xrdp? First solution would be preferable, but second would also work for me

Thanks

Raspberry pi info:
````
lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye
````
````
cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
````
````
uname -r
5.15.76-v8+
````
````
uname -a
Linux pi4 5.15.76-v8+ #1597 SMP PREEMPT Fri Nov 4 12:16:41 GMT 2022 aarch64 GNU/Linux
````

lurch, 2022-12-31:
pinging @maxnet as I believe he's used both Qt5 anf Qt6, so might have an idea why they're behaving differently?

mikey1963, 2023-04-06:
I don't see this problem on bullseye but I do see it on bookworm.