raspberrypi/Raspberry-Pi-OS-64bit #248: Widevine for 64-bit chromium now achievable

Botspot, 2023-02-15:
Until recently, all ARM64 ChromeOS devices were using a 32-bit userland. Google is rolling out an ARM64 userspace in beta for some chromebooks, and a few hours ago an ARM64 Widevine library was successfully snatched from Google's update servers.
See: https://gist.github.com/teacupx/9393507ad6250429707f0318b04f1a3b?permalink_comment_id=4471161#gistcomment-4471161
@theofficialgman is hosting the files [here](https://github.com/theofficialgman/testing/releases/download/gmans-releases/WidevineCdm.tar.gz) for easy access.

I have been able to verify that this libwidevinecdm.so file is indeed in ARM64 format.
```
$ file /tmp/WidevineCdm/_platform_specific/cros_arm64/libwidevinecdm.so
/tmp/WidevineCdm/_platform_specific/cros_arm64/libwidevinecdm.so: ELF 64-bit LSB shared object, ARM aarch64, version 1 (SYSV), dynamically linked, BuildID[sha1]=b71fd3342a03fdb140e1f7a757960df219d63625, stripped
```
This is where we need assistance from @popcornmix, @XECDesign, and anyone else familiar with the current Chromium [patches](https://github.com/RPi-Distro/chromium-browser/tree/master/debian/patches) for Netflix.
Chromium does not seem to detect the Widevine library, despite it being placed in the /opt folder in the exact same manner that the `libwidevinecdm0:armhf` package does. Nobody I have talked to knows enough about Chromium and Glibc to determine if [glibc needs patching](https://thebrokenrail.com/2022/12/31/xfinity-stream-on-linux.html#how-do-i-actually-do-this), chromium needs patching, if they both need patching, or if there is some other foundational obstacle that must be fixed first.

This issue is intended to be center of discussion for the topic of making ARM64 Chromium play Widevine DRM-restricted content. If the Pi team takes this and makes it work in PiOS64, it is my hope that they will disclose what was necessary to make it work, so other browsers, SBCs, and distros can help accelerate the advent of mainstream ARM desktop computing.

XECDesign, 2023-02-16:
Thanks @Botspot, we have got it working internally last week and I'll package it up when I get a chance. Unfortunately things are quite busy at the moment, so I can't say when that might be.

popcornmix, 2023-02-16:
There are two missing symbols in libwidevinecdm.so which prevents it loading.
But using LD_PRELOAD with a suitable stub makes it work.

theofficialgman, 2023-02-16:
please do share what you have done to make it work
In my experience there are multiple issues. 

Firstly the widevinecdm was built to required some GLIBC 2.29+ symbols

Second, on most chromium builds it appears that widevine does not even attempt to load arm arm64. this is the case on debian and ubuntu and multiple ppas and flatpak chromium builds I have tried. the only chromium verison that attempts to load widevine on arm64 seems to be chromium-freeworld from fedora rpmfusion.

Thirdly, on systems with GLIBC 2.36+, running the widevinecdm will trigger an error as the symbol `GLIBC_ABI_DT_RELR` is not present in the widevinecdm library but it calls `DT_RELR` (further explained here <https://thebrokenrail.com/2022/12/31/xfinity-stream-on-linux.html#how-do-i-actually-do-this>). However `DT_RELR` was only added into GLIBC in 2.36 on linux while chromeOS made a specific patch to include it in their own glibc. You can see this with ldd on all systems or on the chromium-freeworld that does attempt to load the binary.
```
[10515:10515:0214/185720.123991:ERROR:content_main_runner_impl.cc(394)] Unable to load CDM /usr/lib64/chromium-freeworld/WidevineCdm/_platform_specific/linux_arm64/libwidevinecdm.so (error: /usr/lib64/chromium-freeworld/WidevineCdm/_platform_specific/linux_arm64/libwidevinecdm.so: DT_RELR without GLIBC_ABI_DT_RELR dependency)
```

I have not been successful in running the binary on any system due to these issues.

Please share any progress.

theofficialgman, 2023-02-16:
ah I see why generic chromium builds do NOT check for widevine

its hardcoded into the widevine.gni script to only enable on `x64` linux https://chromium.googlesource.com/chromium/src/+/HEAD/third_party/widevine/cdm/widevine.gni#23
that explains why all the ppas, debian, and ubuntu chromium builds do work and check for widevine on my x86_64 devices but don't on arm64

tmm1, 2023-02-19:
cc https://github.com/xbmc/inputstream.adaptive/issues/1128

DarkevilPT, 2023-03-19:
How can we properly install this on Chromium now? I'm totally confuse.

popcornmix, 2023-03-19:
I believe it is installed by default on recent images.
You can manually install with `sudo apt install libwidevinecdm0`