raspberrypi/Raspberry-Pi-OS-64bit #250: java 8 not found

TimyIsCool, 2023-04-01:
Im getting
E: Unable to locate package openjdk-8-jdk
E: Unable to locate package openjdk-8-jre
when trying to install java 8 because i need it to do minecraft testing for 1.12 but i cant find any java 8 that works on my pi
running PIOS 64bit on my Pi4 8GB

pelwell, 2023-04-01:
It looks like Debian hasn't supported openjdk-8 since Stretch. Bullseye is on openjdk-11 - is that no good?

TimyIsCool, 2023-04-01:
nope, need java 8

XECDesign, 2023-04-03:
You'll need to download it from Oracle or Azul. We rely on Debian for Java support, so if they don't have it, there's not much we can do.

closed 2023-04-03 by XECDesign

rag2, 2023-04-05:
If you're feeling both very brave & fearlessly motivated, then there is 
an answer, more likely an adventure, to be found at 
https://unix.stackexchange.com/questions/699507/how-can-i-install-java-8-on-debian-11-bullseye. 
*However:* the voice of experience suggests that there may be issues 
with dependencies. Back stuff up first—preferably try this on a second 
copy of your existing OS.

And at that point you are very much taking your own responsibility, 
especially for any unanticipated outcomes.

HTH,

Robin