raspberrypi/Raspberry-Pi-OS-64bit #253: Onboard keyboard not working

adevoss, 2023-05-26:
After a fresh install (2023-05-03 desktop) the onboard keyboard can't be used.

```
user@HOST:~ $ sudo apt install --reinstall onboard
Setup onboard (1.4.1-5+b3) ...
/usr/lib/python3/dist-packages/Onboard/LayoutLoaderSVG.py:447: SyntaxWarning: 's
tr' object is not callable; perhaps you missed a comma?
  raise Exceptions.LayoutFileError("Unrecognized modifier %s in" \
```

```
user@HOST:~ $ sudo onboard
22:58:13.204 WARNING Config: mousetweaks GSettings schema not found, mousetweaks integration disabled.

(onboard:1398): dbind-WARNING **: 22:58:13.427: AT-SPI: Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown: The name org.a11y.Bus was not provided by any .service files

(onboard:1398): dbind-ERROR **: 22:58:13.428: AT-SPI: Couldn't connect to accessibility bus. Is at-spi-bus-launcher running?
```

Is there a way to make this keyboard work?

miguel4tech, 2023-09-04:
I found this useful. https://askubuntu.com/questions/639574/lubuntu-cant-open-onboard-keyboard

So just open the Preferences>Onboard Settings>Typing Assistance - Word suggestions and toggle off (disable) Show suggestions.

Also ensure Auto-show is disabled as well.
That should solve it.
WORKED FOR ME.

adevoss, 2023-09-05:
Worked for me as well.
The Auto-show feature is the reason I prefer this keyboard.
Now there is no difference in using the matchbox and the onboard keyboard. Both need to be made visible manually.

It worked in the past. Don't know what changed to cause onboard not working.
32bit to 64bit OS? buster to bullseye? A change in onboard? All possible.