raspberrypi/Raspberry-Pi-OS-64bit #254: not able to autostart pcp.service

alembiq, 2023-08-31:
I have a problem with a `pcp` package; in my case installed by `apt install cockpit-pcp`.
My issue is that it's not starting after boot, and I cannot make it start by using `systemctl enable pcp.service`, as it doesn't have a systemd service definition.

```bash
systemctl status pcp.service 
● pcp.service - LSB: Legacy control for PCP daemons
     Loaded: loaded (/etc/init.d/pcp; generated)
     Active: inactive (dead)
       Docs: man:systemd-sysv-generator(8)

systemctl enable pcp.service 
pcp.service is not a native service, redirecting to systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable pcp
update-rc.d: error: pcp Default-Start contains no runlevels, aborting.
```

```
apt list --installed | grep pcp

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

cockpit-pcp/oldstable,now 239-1 arm64 [installed]
libpcp-gui2/oldstable,now 5.2.6-1 arm64 [installed,automatic]
libpcp-import1/oldstable,now 5.2.6-1 arm64 [installed,automatic]
libpcp-mmv1/oldstable,now 5.2.6-1 arm64 [installed,automatic]
libpcp-pmda-perl/oldstable,now 5.2.6-1 arm64 [installed,automatic]
libpcp-pmda3/oldstable,now 5.2.6-1 arm64 [installed,automatic]
libpcp-trace2/oldstable,now 5.2.6-1 arm64 [installed,automatic]
libpcp-web1/oldstable,now 5.2.6-1 arm64 [installed,automatic]
libpcp3/oldstable,now 5.2.6-1 arm64 [installed,automatic]
pcp-conf/oldstable,now 5.2.6-1 arm64 [installed,automatic]
pcp/oldstable,now 5.2.6-1 arm64 [installed,automatic]
python3-pcp/oldstable,now 5.2.6-1 arm64 [installed,automatic]
```

2023-08-31 updated raspberry pi os 64bit lite
6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 GNU/Linux

popcornmix, 2023-08-31:
If it is a bug, then it's a likely a debian one rather than PRiOS (it's very unlikely we've modified this package).

popcornmix, 2023-08-31:
[This](https://bugzilla.redhat.com/show_bug.cgi?id=1238120) is possibly relevant.

closed 2023-08-31 by ghollingworth

alembiq, 2023-09-01:
> [This](https://bugzilla.redhat.com/show_bug.cgi?id=1238120) is possibly relevant.

some things probably changed in those eight years, pmlogger is crashing when you ask it for metrics and pcp isn't running