raspberrypi/Raspberry-Pi-OS-64bit #256: Wont load past the broken kernel page

nickeatspie, 2023-09-22:
I bought this thing a few months ago and never got around to using it. even bought a touch screen for it. I want to finally try and figure it out and its showing me this message saying the kernel is broken and wont load past that screen. I cant remember exactly what it says but i know its a common problem. Im wondering if i should just try to return it and hope they send me a working one or is there a fix? Its been a while so i dont think amazon will honor a return at this point. I already had to return the first one i got and still to this day have never gotten to use/ play with it. Any help would be greatly appreciated. thank you. For context its a Raspberry pi 4 16 gig. again thanks

popcornmix, 2023-09-22:
This isn't the right place for support issues like this. Can you make a post in the forum?
https://forums.raspberrypi.com

It would be helpful if you could post the exact error message (or take a photo of the screen).

nickeatspie, 2023-09-22:
thanks ill do that